exelban/stats #1019: Number of processes

lawrence-z-arabii, 2022-08-05:
**Describe the bug**
It's rather a bunch of question than a bugs. Sorry for bad translations of options mentioned below, don't know how to change application language to English instead of system default one, is that even possible?

1) What does "Number of processes" option means in CPU, memory and battery tabs?
2) Am I suppose correct that percentage value of CPU and GPU means its usage?
3) What is the difference between Refresh interval and Refresh of processes interval in CPU and memory tabs?
4) CPU tab seems to be duplicated. Is that for some reason?

Thanks for answers in advance.

exelban, 2022-08-07:
Hi. Thanks for the feedback.

1. It defines the number of processes that are visible in the popup view (the number of top processes visible in the details)
2. Yes
3. Refresh interval determines the interval between reads for the main reader (usage, etc). Refresh of processes interval means how often the top process list will be updated.
4. What?

lawrence-z-arabii, 2022-08-07:
Saying duplicated CPU tabs I meant that:
[image]

Is there any way to switch language version?

exelban, 2022-08-07:
Wow, I don't know how it's possible. I'm not sure it's possible at all.
What mac do you have and what version of Stats do you use?

exelban, 2022-08-07:
Yeap, you can do it with macos location settings.

lawrence-z-arabii, 2022-08-07:
> Wow, I don't know how it's possible. I'm not sure it's possible at all.
> What mac do you have and what version of Stats do you use?

- MacBook Pro M1 Max
- macOS 12.5 (21G72)
- Stats 2.7.28

exelban, 2022-08-07:
Do you have duplicated CPU tabs even after the app restart?

lawrence-z-arabii, 2022-08-07:
You're right. I've just restart app after changing location settings as you mentioned and tab is gone. I didn't hit on with such a obvious solution :-)

exelban, 2022-08-07:
Still, it's strange, I don't know how it's possible to see duplicated tabs.
I will try to fix that if I will find a way to reproduce that.

Thanks for your feedback.

closed 2022-08-07 by exelban