exelban/stats #1022: wrong numbers of CPU efficiency core and performance core

rakexue, 2022-08-08:
**Describe the bug**
wrong numbers of CPU efficiency core and performance core 
Apple M1 has 4 performance and 4 efficiency core.

**Details:**
 - Device: [MacBook Pro (13-inch, M1, 2020) Chip Apple M1 ]
 - macOS: [12.5 ]
 - Application version: [2.7.29]
[image]

BitesPotatoBacks, 2022-08-08:
@rakexue From my knowledge, this inaccuracy is a result of how macOS offers its hardware sensors, not an issue with the Stats app itself.

exelban, 2022-08-08:
hi. @BitesPotatoBacks correctly says. There is no dedicated sensors per core. It's just a sensors that does not connected to specific cores.

closed 2022-08-08 by exelban