exelban/stats #1024: [FR] Show low power mode when on

Skee, 2022-08-08:
Hi,

I replaced the native battery menu item with the Stats one and it's very nice, but I'm missing the original's feature of showing when the mac is in low power mode.

Could a "Low Power Mode" status entry be included in the detail view, either of the Battery widget (similar to original battery view behaviour) or to the CPU widget (which also makes sense)?

Original behavior is to only show the line item when low power mode is on.

Thanks.

exelban, 2022-08-08:
Hi. Thanks for the feedback.
Yeap, it could be a nice feature. I will check if it's possible to obtain this info.

ffffwh, 2022-08-18:
It is acceptable that
- providing a shortcut to toggle LPM in app
- not guaranteeing accurate LPM status, if it is changed else where

closed 2022-08-21 by exelban

Skee, 2022-08-21:
Many thanks for this!

wzxu, 2022-09-01:
@exelban Just wondering if toggling on/off LPM is implemented? I only see a read-only line of text on the popover. Am I missing something? Thanks.

exelban, 2022-09-01:
@wzxu nope, it will be not implemented.

ffffwh, 2022-09-01:
The difficulties might be that a PrivilegedHelperTools is required to toggle LPM.

It is important to use LPM to control temperature on a fan-less Mac, just like adjusting fan speed in this app for a fan-equipped Mac.

My Air was running at ~105℃ for the whole night due to a faulty program. It would be great if LPM was turned on automatically on high temperature.

I don't trust Apple's value for throttling temperature. The 12-inch MacBook is infamous for CPU solder failure due to high temperature.

wzxu, 2022-09-01:
> @wzxu nope, it will be not implemented.

Got it 👌