exelban/stats #1025: Option to adjust scaling manually

Omoeba, 2022-08-08:
I noticed "feat: added different scaling types to the charts (none, square, cube, logarithmic)" in the `v2.7.29` release notes but there doesn't seem to be any way to adjust that in the UI. It would be great if you could set the scaling manually, for example to the sensors graphs.

exelban, 2022-08-09:
Hi. Yeap, this feature is already in the app.
You can manually set the scale for network chart and line chart widgets.
For now, it's available only for widgets, I'm not decided if it will be added to the charts in the popup view.

closed 2022-08-09 by exelban

Omoeba, 2022-08-10:
I should've specified that I'm asking about the charts since it is really difficult to see spikes in something like this.
[image]