exelban/stats #1027: Stats can't display all resource usage on the host when not run as 'root', or with higher privileges.

boldandbusted, 2022-08-09:
**Describe the bug**
Stats can't display all resource usage on the host when not run as 'root', or with higher privileges.

**Details:**
If a user wishes to see how all applications and programs are utilizing resources on the Mac, they must elevate their privileges to do so. While this might seem trivial, all it takes is one 'security' daemon to ruin the usefullness of your Mac. ;) I noticed this when my fans start speeding up but Stats shows little or no CPU or storage activity.

Illustration:
```shell
$ sudo ps auwx | wc
Password:
     539    7949  129536

$ ps auwx | wc
     537    7919  129188

py-3.8.13 AHD-MB-C02C10EWLVDM in ~
○ →
```

In this case, there are two extra processes only visible when run as root.

Is there a way to easily have Stats run with enough privileges to see all resource usage, but without sacrificing other Stats features (like notifications, automatic start on user account login, etc.)?

I love Stats, and hope this helps improve it. :) Cheers! :)

exelban, 2022-08-11:
Hi. Thanks for the feedback.
I'm not sure I understand the problem correctly. 
Do you want to run a Stats as a root? Or do you mean that the Stats does not show the processes that were run as a root?

exelban, 2022-08-20:
Hi. Going to close the issue because I have no idea what is issue about.

closed 2022-08-20 by exelban