exelban/stats #1029: Incorrect RAM usage

lawrence-z-arabii, 2022-08-10:
**Describe the bug:**
RAM tab shows incorrect usage.

**Details:**
 - Device: Macbook Pro M1 Max 32 cores 32GB
 - macOS: 12.5.0
 - Application version: 2.7.28

Hi, I was happened the issue as below.

AutoCAD LT 2023 process almost always consumes over 300 GB:
[image]

Restarting Stats app doesn't change anything. However when I restart AutoCad app, RAM seems to be shown let's say correctly:
[image]

Not so long though. No matter what, it always rises up to over 300 GB. From my observations at the beginning growth is stepped, but at some point RAM suddenly starting to show false high value. Last true (I guess) value I've noticed was about 7,7 GB after 1 hour of working.

exelban, 2022-08-10:
Hi. Thanks for your feedback.
Could you please share the output of `top -l 1 -o mem -n 8 -stats pid,command,mem` when you will see 300GB?

It's not a wrong usage, but wrong `top` output parsing.

exelban, 2022-08-10:
And now the Stats does not show correct values in the process section?

exelban, 2022-08-10:
thanks, will fix that.

closed 2022-08-14 by exelban