exelban/stats #1034: [Feature Request] Battery indicator with percentage inside icon

sampl3x, 2022-08-14:
Is it possible to make a battery indicator with the percentage inside the icon which is coming in macOS Ventura? 
This save space in the Menubar and older mac who don't have Ventura support can have it to.

[image]

exelban, 2022-08-14:
Yes, it could be implemented.

closed 2022-08-28 by exelban

sampl3x, 2022-08-29:
Thank you for adding this. Is it possible to make the battery icon bigger/higher? The text inside the battery is barely readable/very tiny.

Atemu, 2022-08-29:
I've also noticed that the battery doesn't shrink anymore. How does that work in Ventura?