exelban/stats #1052: [Feature Request] Add Battery Max / Hottest Battery to temperature sensors

richardzone, 2022-08-23:
I'd like to propose a `Battery Max` sensor similar to what we already have("Hottest CPU"). Some reasoning:

1. Although high temperature is bad for all electronic components, battery suffers the most from this.
2. Macs Fan Control app already have this as part of the sensors.

exelban, 2022-08-24:
Hi. Thanks for the feedback.
Do you have multiply batteries in your mac?

richardzone, 2022-08-24:
Hi @exelban  thanks for quick reply! I have only one battery built-in with my mac. 

For `Battery Max` sensor I mean it would be the higher value of either `Battery` or `Battery 2`

exelban, 2022-08-24:
It has no sense. In the case of CPU or GPU, it is just the hottest core. In case of battery 1 and 2 it's just different sensors. And don't represent the same. For example, battery 1 could be the battery temperature, and battery 2 is a BMS sensor. And it cannot be merged.

closed 2022-08-24 by exelban