exelban/stats #1053: App is Repeatedly asking for elevated permissions via “Apple wants to make changes”.

ShlomoStept, 2022-08-24:
Describe the bug:
The stats app has repeatedly asked for elevated privileges "to make changes”,
I'm not sure if this means the app is secretly malware, or maybe its being hijacked by some other malware, but i just wanted to see if anyone else is having this issue

Details:
 - Device: MacBook M1 Pro
 - macOS: 14.4.0
 - Application version: 2.7.31

exelban, 2022-08-24:
Hi. Thanks for the feedback.
Do you change the fan speed? If yes, it requires root access. But it asks only once.
And the message must probably will be `Stats wants to make changes`.

ShlomoStept, 2022-08-25:
I don't manually change the fan speed.
Not sure if this information helps, but i have had this app for over 2 months now and this is a new problem that happened several times in the last week, 

I guess no-one else had this problem. I'm going to delete and reinstall it, and ill let you know if this continues to happen

exelban, 2022-08-25:
In case you don't change the speed, there is no way that Stats asks for the root. It's only one function that requires root access.
Also, I'm pretty sure that it's not Stats asks for root. Because it must be written that Stats asks for that.

rekonnected, 2022-08-25:
Also having this issue. No fan control of any kind enabled or used unless it's on by default, but it only started asking for elevation recently and I have been rejecting it as well for the same reasons as @ShlomoStept, especially since Stats doesn't give a explanation/reason for requesting elevation meaning I don't know what it's doing.

macOS Monterrey 12.5.1 (21G83)
Stats 2.7.31
All Fan-relateed settings disabled as shown in attached picture. 
[image]

exelban, 2022-08-25:
Please share the screenshot of the alert.

rekonnected, 2022-08-25:
It has only happened occasionally and inconsistently on my machine, so I don't know if I can manually trigger it. 

It's a macOS system modal so it should be the same for just about all apps requesting elevation, just replace the icon and app name with Stats: 
[image]

One thing that might be a good change for the future would be if something like what Dropbox did in this example could be implemented, giving some custom text to explain the reason for elevation: 
[image]

exelban, 2022-08-25:
Please share the screenshot of the sensors module popup window.

exelban, 2022-08-25:
thanks.
It's not possible that Stats will ask for the changes if you do not manipulate the fans. It's quite complicated to show this window. And there is one place when it shows. 
Maybe try disabling the modules, one by one, to see when it stops showing this window. And try to reinstall the app.

SamadiPour, 2022-08-25:
I hope this helps.
All modules are disabled except network.

https://user-images.githubusercontent.com/24422125/186734603-56ce13f7-05e4-444f-ab0e-78716ebf3041.mov

My stats menu:
[image]

exelban, 2022-08-25:
And what if you disable the network module but leave enabled the CPU module?

exelban, 2022-08-25:
Does it present for some time, or does it starts after the last update?

exelban, 2022-08-25:
Does the window appear with an empty username?

Garretsw, 2022-08-25:
I have recently had this issue and I do not believe it started until the last update @exelban . Mine had the username prefilled and so far has only showed once since I did approve the extra permissions.

denizt, 2022-08-26:
I didn't want to post a me-too post but this happens to me, too. I would normally debug and fix this myself and follow up with a PR but it's quite busy these days... I did some debugging which might help with the investigation.

When I run Stats from the terminal, it will try to elevate but not show the popup. The log is:

```
cannot upgrade owner to root: Optional({
    NSAppleScriptErrorAppName = Stats;
    NSAppleScriptErrorBriefMessage = "The administrator user name or password was incorrect.";
    NSAppleScriptErrorMessage = "The administrator user name or password was incorrect.";
    NSAppleScriptErrorRange = "NSRange: {0, 187}";
})
```

The only source for this I can find is in https://github.com/exelban/stats/blob/7d35e0f9897cc1708b874a9b3b3f46ec8f846d29/Kit/helpers.swift#L1054

It happens on wake up from sleep, let me know if you can't reproduce and want me to debug locally, I will try to find some time during the weekend.

Also to be on the side of err, it would be good to make sure the release is not tampered with, highly unlikely but with so many users using stats, it would probably be better safe than sorry.

y2ducky, 2022-08-26:
The same problem is happening to me and it's annoying. I would appreciate it if you could correct this problem or revert to ㅍv2.7.30.

exelban, 2022-08-26:
@Garretsw yeap, I think you can be right. Will check that.

exelban, 2022-08-26:
Yeap, it was fan reset. There was no check if the fan mode was changed. And when mac goes to sleep mode, the Stats tried to reset the fans. The fix will be in the new release later this week. There is a build with this fix, you can use it till then.

[Stats.dmg.zip](https://github.com/exelban/stats/files/9432300/Stats.dmg.zip)

ShlomoStept, 2022-08-26:
@exelban Ok great, thanks for fixing this so quickly!

Also, I don't know the protocol, should I close the issue now, or wait until the next release?

exelban, 2022-08-26:
I will close the issue when the release with that fix will be available. To prevent duplicating.
Thanks for the feedback.

closed 2022-08-28 by exelban

infinitepower18, 2022-09-11:
I faced the same problem few weeks ago. How can I undo the permission changes? I only use this for RAM and network monitoring.