exelban/stats #1054: What is supposed to happen to the menu barwhen you install stats?

sagnik, 2022-08-25:
This is a dumb question and I apologize in advance. I installed stats, restarted the machine and I was expecting the menu bar to change. Is that a correct expectation? My menu bar has no changes and stats is installed as an application so I am totally confused. May be I am understanding this wrong?

exelban, 2022-08-25:
Hi. What mac do you have?

sagnik, 2022-08-25:
[image]

exelban, 2022-08-25:
There is no enough space in the menu bar. You need to close some apps in the menu bar to see new widgets.

closed 2022-08-25 by exelban

rekonnected, 2022-08-25:
@sagnik Try [Hidden Bar](https://github.com/dwarvesf/hidden) (free) or [Bartender](https://www.macbartender.com) (paid) to help hide and manage menu bar icons.

sagnik, 2022-08-25:
That was it. closing. Thanks for the help!