exelban/stats #1061: Missing translation Strings

Jerry23011, 2022-08-30:
Hi, I'm a Chinese user of Stats and really want to say thank you for creating such an amazing app!
I just PR last week to improve the Simplified Chinese Translations and I find that there are some missing translations that don't have corresponding strings in the Localizable.string file. 
For example, the "Current" and "Energy" in the sensors interface doesn't have the strings ready for us to make translations. I noticed the same issue with the French version as well.
[image]
Could you create those strings in the Localizable.strings files so that we can PR and make the translations? Or maybe we can add those words ourselves when we PR?

exelban, 2022-08-30:
Hi. Thanks, I didn't notice. It will be fixed.

Jerry23011, 2022-08-30:
Thank you very much! I'll get the Simplified Chinese Translation done as soon as you add the strings😄

Jerry23011, 2022-08-30:
BTW, there are other parts missing translations, too. Here's the list.:
1. The pop up menu of GPU(in the menu bar), "Cores" is missing[image]

2. The unit of the update interval "sec"
[image]

3. The "Autodetection" option in Network
[image]

4. The "Percentage inside icon" option in the Widget settings in Battery
[image]

closed 2022-09-04 by exelban