exelban/stats #1063: Network -> Public IP not showing

valynor, 2022-09-01:
Stats 2.7.32 on Mac Studio M1 Max, macOS 12.5.1

The Network popup is not showing my IP properly:

v4: just says "Unknown"
v6: the first half of the address shown is not my IP but my IPv6-prefix /56 .. and I actually don't know what the second half is, but it's not my IPv6 IP as shown in my router 

I'm on a Dual Stack v4/v6 VDSL connection behind a NAT router.

exelban, 2022-09-01:
Hi. Thanks for the feedback.
Do you see your IP [here](https://api.serhiy.io/v1/stats/ip)?

exelban, 2022-09-01:
So it means that you don't have an ipv4 or it's not available. It's not something that could be obtained in the system. Only via a remote server. Prefix it's just a subnet mask.

closed 2022-09-01 by exelban

valynor, 2022-09-01:
I understand the v6 problem but sites like https://www.whatismyip.com/ or https://www.deviceinfo.me/ show my IPv4 correctly.