exelban/stats #1064: Missing Bluetooth Device Stats

leobrafl, 2022-09-01:
Hello,
My apple mouse is not properly showing its battery life on the menu bar as suggested by stats. I'll attach a print screen (covering my name) of it to show. I know it's on 59% via the control center, but it doesn't show it via stats
A clear and concise description of what the bug is.

**Details:**
 - Device: M1 MAX Macbook Pro 16 2021
 - macOS: 12.5.1
 - Application version: 2.7.32
 - Apple Mouse 1

Images

[image]
[image]

exelban, 2022-09-02:
Hi. Thanks for the feedback.
Please share the output of `system_profiler SPBluetoothDataType -json`. But please find only the part where your headphones are, and share only this part. Also, remove the sensitive data (serial number and address) from it.

caiquebb, 2022-09-06:
Same here!

Here is the JSON:
```json
{
  "SPBluetoothDataType" : [
    {
      "controller_properties" : {
        "controller_address" : "14:7D:DA:4E:47:D2",
        "controller_chipset" : "BCM_4364B3",
        "controller_discoverable" : "attrib_off",
        "controller_firmwareVersion" : "v86 c4205",
        "controller_productID" : "0x0001",
        "controller_state" : "attrib_on",
        "controller_supportedServices" : "0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >",
        "controller_transport" : "UART",
        "controller_vendorID" : "0x004C (Apple)"
      },
      "device_connected" : [
        {
          "Magic Keyboard with Numeric Keypad" : {
            "device_address" : "90:9C:4A:0A:C2:E4",
            "device_firmwareVersion" : "1.6.0",
            "device_minorType" : "Keyboard",
            "device_productID" : "0x026C",
            "device_services" : "0x800020 < HID ACL >",
            "device_vendorID" : "0x004C"
          }
        },
        {
          "Magic Mouse 2" : {
            "device_address" : "A8:91:3D:36:B7:F9",
            "device_batteryLevelMain" : "100%",
            "device_firmwareVersion" : "1.9.2",
            "device_minorType" : "Mouse",
            "device_productID" : "0x0269",
            "device_services" : "0x800020 < HID ACL >",
            "device_vendorID" : "0x004C"
          }
        }
      ],
      "device_not_connected" : [
        {
          "Caique’s AirPods Pro" : {
            "device_address" : "20:78:CD:4B:36:29",
            "device_firmwareVersion" : "4E71",
            "device_minorType" : "Headphones",
            "device_productID" : "0x200E",
            "device_serialNumber" : "H7CH2JC31059",
            "device_vendorID" : "0x004C"
          }
        },
        {
          "Caique’s iPhone" : {
            "device_address" : "14:D1:9E:C4:12:4D"
          }
        },
        {
          "JBL T450BT" : {
            "device_address" : "78:44:05:BB:DC:69",
            "device_firmwareVersion" : "0.0.0",
            "device_minorType" : "Headset"
          }
        }
      ]
    }
  ]
}
```

exelban, 2022-09-06:
@caiquebb please share the screenshot of the Battery module popup

leobrafl, 2022-09-06:
Hello, sorry for the delay, but here it is:
I've already shared the Battery Module Popup

{

  "SPBluetoothDataType" : [

    {

      "controller_properties" : {

        "controller_address" : "F4:D4:88:6A:76:7A",

        "controller_chipset" : "BCM_4387",

        "controller_discoverable" : "attrib_off",

        "controller_firmwareVersion" : "19.6.451.5119",

        "controller_productID" : "0x0001",

        "controller_state" : "attrib_on",

        "controller_supportedServices" : "0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >",

        "controller_transport" : "PCIe",

        "controller_vendorID" : "0x004C (Apple)"

      },

      "device_connected" : [

        {

          "Apple Mouse" : {

            "device_address" : "34:15:9E:D3:3B:1E",

            "device_firmwareVersion" : "0.8.4",

            "device_minorType" : "Mouse",

            "device_productID" : "0x030D",

            "device_services" : "0x800020 < HID ACL >",

            "device_vendorID" : "0x05AC"

          }

        }

      ],

      "device_not_connected" : [

        {

          "Galaxy Buds (AB72)" : {

            "device_address" : "CC:21:19:BF:AB:72",

            "device_firmwareVersion" : "0.0.1",

            "device_minorType" : "Headset",

            "device_productID" : "0xA011",

            "device_vendorID" : "0x0075"

          }

        },

        {

          "JBL Charge 3" : {

            "device_address" : "B8:D5:0B:B3:A8:72",

            "device_firmwareVersion" : "0.0.0",

            "device_minorType" : "Speaker",

            "device_productID" : "0x0023",

            "device_vendorID" : "0x0057"

          }

        },

        {

          "Apple Mouse" (This is broken though) : {

            "device_address" : "04:0C:CE:3C:D2:B8",

            "device_firmwareVersion" : "3.0.6",

            "device_minorType" : "Mouse",

            "device_productID" : "0x030D",

            "device_vendorID" : "0x05AC"

          }

        },

        {

          "S21+ de Leonardo" : {

            "device_address" : "60:68:4E:1A:52:59",

            "device_firmwareVersion" : "2.0.1",

            "device_minorType" : "MobilePhone",

            "device_productID" : "0x0100",

            "device_vendorID" : "0x0075"

          }

        }

      ]

    }

  ]

}%

exelban, 2022-09-06:
@leobrafl thanks, unfortunately this device does not report the battery level. As you can notice in the `Mouse de ****** ` section there is no level.

leobrafl, 2022-09-06:
Well, just testing with my other mouse, the one I mentioned is broken, and iStats is recognizing it just fine. Maybe they are getting this info from another source? I ran the system_profiler SPBluetoothDataType -json again and, again, there was no signs of battery over there, so they are definitely getting that info from another source

```
{
  "SPBluetoothDataType" : [
    {
      "controller_properties" : {
        "controller_address" : "F4:D4:88:6A:76:7A",
        "controller_chipset" : "BCM_4387",
        "controller_discoverable" : "attrib_on",
        "controller_firmwareVersion" : "19.6.451.5119",
        "controller_productID" : "0x0001",
        "controller_state" : "attrib_on",
        "controller_supportedServices" : "0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >",
        "controller_transport" : "PCIe",
        "controller_vendorID" : "0x004C (Apple)"
      },
      "device_connected" : [
        {
          "Apple Mouse" : {
            "device_address" : "04:0C:CE:3C:D2:B8",
            "device_firmwareVersion" : "3.0.6",
            "device_minorType" : "Mouse",
            "device_productID" : "0x030D",
            "device_services" : "0x800020 < HID ACL >",
            "device_vendorID" : "0x05AC"
          }
        }
      ],
```

You'll see the firmware in this one is the one that I mentioned it was broken

exelban, 2022-09-06:
Unfortunately there I know only this way for Bluetooth battery level. Give me know if someone has an idea how it can be fetched in another way.

closed 2022-09-06 by exelban