exelban/stats #1068: Invisible Bluetooth

nikhilbadyal, 2022-09-03:
Unable to see the Bluetooth icon in the menu bar. It's transparent
[image]

exelban, 2022-09-04:
Yeap, it's because there are no devices to show.

closed 2022-09-04 by exelban

nikhilbadyal, 2022-09-04:
Thats not true. Even with device its invisible. 
[image]

exelban, 2022-09-04:
Yeap, but that device does not have a battery level.

nikhilbadyal, 2022-09-05:
That's not true. My device does have a battery level indicator.
[image]