exelban/stats #1080: Network Chart does not remember color settings

mhtawfiq, 2022-09-11:
**Describe the bug**
After setting the color of download and upload fr the Network Cart widget, when you restart Stats the color settings are forgotten and the chart uses the default colors again.

**Details:**
 - Device: Macbook Pro 2016
 - macOS: 10.15.1
 - Application version: 2.7.34

exelban, 2022-09-11:
Hi. Thanks for reporting.
As I can notice it looks like the colors are saved. But reversed (upload and download). I cannot reproduce that colors are reset.

mhtawfiq, 2022-09-12:
Hi

I just double checked and your observation is correct. The download/upload colors are saved but reversed on Stats app restart.

Thank you.

exelban, 2022-09-12:
Thanks. Will be fixed.

closed 2022-09-18 by exelban

mhtawfiq, 2022-09-18:
Thank you! I just tested 2.7.35 and the issue is fixed now.