exelban/stats #1081: Won't properly upgrade when running as a Standard User

kalaklanar, 2022-09-12:
When running as a standard user, and propmpted to install a new version, the exe in /Applications is no longer useable/is corrupted.

Observered:
1. Stats running
2. prompted to download new version
3. click Download
4. Click Install
5. /Applications/Stats.app is no longer executable (🚫 icon shown)

Expected:
Either will not offer to install (will just download) if not an admin or will ask for admin credentials and stop update if not credentials successfully entered.

kalaklanar, 2022-09-12:
Note: These are both M1 and Intel Macs I've seen it on.

exelban, 2022-09-12:
I already have seen this behavior, and it's not a Stats bug. And cannot be fixed from the app level.
And also the update process does not require root access. As the app itself. The root access only required if you want to control the fans.

closed 2022-09-12 by exelban