exelban/stats #1082: [Feature Request] Add network connectivity icon

anQluck, 2022-09-13:
Please add an option to show resource(IP/DNS) ping status as an green(all OK) / yellow (connected but no ping response) / red (disconnected) style dot icon like in iStat Menus software

exelban, 2022-09-13:
Hi. Thanks for the feedback.
Yeap, I think it will be a nice feature. I will try to do that.
But I don't like the idea of ping with interval some external resources. So this feature will depend if I can find another solution.

anQluck, 2022-09-13:
Sure. Ping an IP is simplest one solution IMHO. I used 8.8.8.8 DNS for ping target. Confident that somewhere in a system exist some built-in status for network status, that can be parsed. So this one will be great realization!

closed 2022-09-25 by exelban