exelban/stats #1084: Combined icon to save on statusbar spacing

kim0, 2022-09-14:
**Describe the bug**
Thanks a million for this great app, which has been running 24x7 on my macos for years. I came here with the intention of filing a request to reduce the icon spacing, since the status bar space is a very rare resource :) I found many done so before me, and found that Apple does not allow this spacing to be adjusted. So what about having a combined icon mode that saves a lot on required statusbar space needed? This can be a special mode for those who care about saving space. For example I only care about CPU/RAM/Network and think those 3 can fit in roughly the same space that shows network activity today.

For example, today network activity has units of KB/s. I think in the space saving mode, we can drop the "B/s" since it's inferred and only show K or M. For the CPU and RAM, I would not mind 2 small numbers on top of each other like how the network shows up/down speeds. It might also be possible to make a single wide icon (like how Apple's VPN icon shows a wide timer displaying how long the VPN has been connected!) and just fill it with the content of the 3 icons.

I understand this is probably not a small change to the code. And will need adjustments to how the popup menu happens, but it's likely a needed change, bec the pain is real :) Many reports have been opened about this topic. So I'll just leave this here. Again, thanks a million!

**Details:**
 - Device: Apple M1 Pro
 - macOS: 13.0 Beta (22A5342f)
 - Application version: 2.7.34

exelban, 2022-09-16:
Hi. Thanks for the feedback.
As I said a lot of times. It will be implemented. But it requires a lot of changes. I don't know when I will find time to do that because of that. I'm pretty sure that it will be the next year, at least. It's a reason why I'm closing the issue about this feature. Because I don't have any time range for that feature.

But I like how this issue is written, so I leave that one.

exelban, 2023-02-26:
Hi. So, for now, this feature is finished. I will have a lot of tweaks to the popup view and add missing modules to the popup view with time.

closed 2023-02-26 by exelban