exelban/stats #1088: The description of the power sensor for the Mac Pro (2013) is missing

molekulek, 2022-09-17:
**Describe the bug**
The description of the power sensor for the Mac Pro (2013) is missing.

**Details:**
 - Device: Mac Pro 2013
 - macOS: 12.6.0
 - Application version: 2.7.34

[image]

exelban, 2022-09-17:
Description of what?

molekulek, 2022-09-17:
No readouts of power values, all of them indicate zero.

exelban, 2022-09-17:
The sensors are from the SMC. Zeros mean that SMC does not return any value.

closed 2022-09-17 by exelban

molekulek, 2022-09-17:
I am asking because iStatMenu is showing those values.

[image]
[image]