exelban/stats #1091: Bluetooth monitor

chenshanb3, 2022-09-19:
Hello, I am very happy to have such a software that can visually check the system status. I would like to make a suggestion. When I was using it recently, I found that this Bluetooth monitor can only display a single power level. I hope it can display all the power levels. It displays 2 lines by default. If there are more than two, I hope it can be scrolled. Thanks!

[image]

exelban, 2022-09-19:
Please ensure that you have the latest version of the app.

chenshanb3, 2022-09-20:
[image]

latest version of the app

exelban, 2022-09-20:
Hmm, looks like #1083 is not fixed. Will take a look again.

chenshanb3, 2022-09-20:
The keyboard and mouse identifiers are recognized as one

[image]

exelban, 2022-09-20:
The bug is in the popup view. It does not remove the missing level.

exelban, 2022-09-20:
update: the problem is not in the popup view. But in the Sensors widgets. The bug appears after sensor widgets receive the option to sort the items. And it uses the key as id. And battery level keys could be repeated in many devices. Will be fixed.

closed 2022-09-25 by exelban