exelban/stats #1093: Allow storing of settings / custom configuration in a file and export it

rushiagr, 2022-09-23:
**Describe the bug**
So that when I use a new mac, I don't have to set it up all over again

**Details:**
 - Device: Macbook Pro 2021 model, 16 inch, M1
 - macOS: Mac OS Monterey
 - Application version: Can't figure out

exelban, 2022-09-23:
Hi. Thanks for the feedback.
I have no plan to add this feature to the app. There are a few reasons why not. The first one is simple: it requires rewriting the all logic that could be saved in the app (there are already a lot of them). The second one is that you can do export/import just with export/import user defaults. There was a ticket where someone described how it could be done.

closed 2022-09-23 by exelban

rushiagr, 2022-09-24:
Thank you, and thanks for the reference to an older ticket. I was able to find it ([link](https://github.com/exelban/stats/issues/801#issuecomment-1020517746)), and this is good enough for me for now.