exelban/stats #1094: GPU stats missing on M1 MBP

Mesmer88, 2022-09-23:
GPU stats are not showing on M1 MBP

 - Device: MacBook Pro 16 M1 2021
 - macOS: Monterey 12.6
 - Application version: 2.7.35

Output of `ioreg -r -d 1 -w 0 -c "IOAccelerator"`

```
+-o AGXAcceleratorG13X  <class AGXAcceleratorG13X, id 0x10000087c, registered, matched, active, busy 0 (17 ms), retain 24>
    {
      "AGXParameterBufferMaxSizeEverMemless" = 574881792
      "IOMatchedAtBoot" = Yes
      "vendor-id" = <6b100000>
      "GPURawCounterBundleName" = "AGXGPURawCounterBundle"
      "MetalStatisticsName" = "AGXMetalStatisticsExternalA14X"
      "GPURawCounterPluginClassName" = "AGXGPURawCounterSourceGroup"
      "MetalPluginClassName" = "AGXG13XDevice"
      "SCMVersionNumber" = ""
      "AGCInfo" = {"fLastSubmissionPID"=390,"fSubmissionsSinceLastCheck"=0,"fBusyCount"=0}
      "MetalPluginName" = "AGXMetalG13X"
      "IONameMatched" = "gpu,t6000"
      "PerformanceStatistics" = {"TiledSceneBytes"=1572864,"Alloc system memory"=1214005248,"Tiler Utilization %"=2,"recoveryCount"=0,"Renderer Utilization %"=2,"SplitSceneCount"=0,"Allocated PB Size"=17956864,"Device Utilization %"=2,"In use system memory"=340983808}
      "IOGLBundleName" = "AppleMetalOpenGLRenderer"
      "AGXParameterBufferMaxSizeNeverMemless" = 287440896
      "IOGLESBundleName" = "AppleMetalGLRenderer"
      "IOSourceVersion" = "190.22"
      "IOPersonalityPublisher" = "com.apple.AGXG13X"
      "IOPowerManagement" = {"CurrentPowerState"=1,"CapabilityFlags"=2,"MaxPowerState"=1,"DriverPowerState"=1}
      "model" = "Apple M1 Pro"
      "AGXTraceCodeVersion" = "3.24.4"
      "SCMBuildTime" = ""
      "CFBundleIdentifier" = "com.apple.AGXG13X"
      "gpu-core-count" = 16
      "IOProviderClass" = "AppleARMIODevice"
      "AGXParameterBufferMaxSize" = 862322688
      "IONameMatch" = ("gpu,t8015","gpu,t8027","gpu,t8030","gpu,t8103","gpu,t6000","gpu,t6001","gpu,t6002")
      "IOReportLegendPublic" = Yes
      "IOClass" = "AGXAcceleratorG13X"
      "CFBundleIdentifierKernel" = "com.apple.AGXG13X"
      "GPUConfigurationVariable" = {"gpu_gen"=13,"num_gps"=8,"num_cores"=16,"core_mask_list"=(255,255),"num_frags"=16,"num_mgpus"=2}
      "IOGeneralInterest" = "IOCommand is not serializable"
      "IOReportLegend" = ({"IOReportChannels"=((1,6442450945,"Alloc system memory"),(2,6442450945,"In use system memory"),(3,6442450945,"GPU Restart Count")),"IOReportGroupName"="Internal Statistics","IOReportChannelInfo"={"IOReportChannelUnit"=0}},{"IOReportChannels"=((76202472785753,4295098369,"GPU Energy")),"IOReportGroupName"="Energy Model","IOReportChannelInfo"={"IOReportChannelUnit"=216173288919924736}},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((387877586224,6442450945,"GPU Power Zone Filter 0"),(387877586225,6442450945,"GPU Power Zone Filter 1"),(387877586226,6442450945,"GPU Power Zone Filter 2"),(387877586227,6442450945,"GPU Power Zone Filter 3"),(387877586228,6442450945,"GPU Power Zone Filter 4"),(77280627148114,6442450945,"GPU Power Filter")),"IOReportChannelInfo"={"IOReportChannelUnit"=7205759403792793600},"IOReportSubGroupName"="Power Zone Filters"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((6072344679969803520,4295098369,"Average Sum"),(6072344679700911104,4295098369,"Average Sum Count"),(6072344679868354048,4295098369,"Min"),(6072344679867832320,4295098369,"Max"),(5572170264793329716,4295098369,"Tg4a Sum"),(5572170264797720628,4295098369,"Tg4a Sum Count"),(5572163616200732724,4295098369,"Tg4a Min"),(5572163582008766516,4295098369,"Tg4a Max"),(5572170264793329719,4295098369,"Tg7a Sum"),(5572170264797720631,4295098369,"Tg7a Sum Count"),(5572163616200732727,4295098369,"Tg7a Min"),(5572163582008766519,4295098369,"Tg7a Max"),(5572170264793329970,4295098369,"Tg12a Sum"),(5572170264797720882,4295098369,"Tg12a Sum Count"),(5572163616200732978,4295098369,"Tg12a Min"),(5572163582008766770,4295098369,"Tg12a Max"),(5572170264793329973,4295098369,"Tg15a Sum"),(5572170264797720885,4295098369,"Tg15a Sum Count"),(5572163616200732981,4295098369,"Tg15a Min"),(5572163582008766773,4295098369,"Tg15a Max")),"IOReportChannelInfo"={"IOReportChannelUnit"=10},"IOReportSubGroupName"="Temperature"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((306290446408,68719607810)),"IOReportChannelInfo"={"IOReportChannelUnit"=72058115876454424},"IOReportSubGroupName"="GPU Performance States"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((4779256294837407816,68719607810)),"IOReportChannelInfo"={"IOReportChannelUnit"=72058115876454424},"IOReportSubGroupName"="GPU Boost Controller Performance States"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((77263479784786,8590065666)),"IOReportChannelInfo"={"IOReportChannelUnit"=72058115876454424},"IOReportSubGroupName"="Fender State"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((20073050495403080,47244771330)),"IOReportChannelInfo"={"IOReportChannelUnit"=72058115876454424},"IOReportSubGroupName"="DVD Request States"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((5138701042469065805,73014575106)),"IOReportChannelInfo"={"IOReportChannelUnit"=72058115876454424},"IOReportSubGroupName"="CLTM-induced GPU Performance States"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((20073050445935688,231928365059,"GPU Active Time")),"IOReportChannelInfo"={"IOReportChannelConfig"=<60090000000000000000000004000000e02e000000000000010000002200000080730c00000000000200000010000000>,"IOReportChannelUnit"=72058115876454424},"IOReportSubGroupName"="Active Time Histogram"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((20073050580216904,231928365059,"GPU Idle Time")),"IOReportChannelInfo"={"IOReportChannelConfig"=<60090000000000000000000004000000e02e000000000000010000002200000080730c00000000000200000010000000>,"IOReportChannelUnit"=72058115876454424},"IOReportSubGroupName"="Idle Time Histogram"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((5066061750913430600,231928365059,"Fender Active Time")),"IOReportChannelInfo"={"IOReportChannelConfig"=<60090000000000000000000004000000e02e000000000000010000002200000080730c00000000000200000010000000>,"IOReportChannelUnit"=72058115876454424},"IOReportSubGroupName"="Fender Active Time Histogram"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((5066061751047711816,231928365059,"Fender Idle Time")),"IOReportChannelInfo"={"IOReportChannelConfig"=<60090000000000000000000004000000e02e000000000000010000002200000080730c00000000000200000010000000>,"IOReportChannelUnit"=72058115876454424},"IOReportSubGroupName"="Fender Idle Time Histogram"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((24020220114194004,4295098369,"UVWarn Instances"),(6149174098653697605,4295098369,"UVWarn Active")),"IOReportChannelInfo"={"IOReportChannelUnit"=7205759403792793600},"IOReportSubGroupName"="UV Warn State"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((5928236105841398868,4311744513,"Total Restarts"),(5928236105841398869,4311744513,"Unknown Restarts"),(5571873383924453956,4311744513,"mediaserverd Restarts"),(5932177713106399795,4311744513,"36223"),(5932177717317873977,4311744513,"41819"),(5932177717334651442,4311744513,"42822"),(75098766265420,4311744513,"DM Kill Count"),(20077427183668300,4311744513,"GTP Kill Count"),(19793714528996428,4311744513,"FRG Kill Count"),(18933922205879372,4311744513,"CDM Kill Count"),(21749771400202572,4311744513,"Memless Render Failed")),"IOReportChannelInfo"={"IOReportChannelUnit"=7205759403792793600},"IOReportSubGroupName"="Issue Counts"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((5139821406163519308,4311744513,"GTP Single"),(5139821406162535760,4311744513,"GTP Dependent"),(5139821406162863180,4311744513,"GTP Idle"),(5139821406162340940,4311744513,"GTP Alloc New"),(5139821406163455556,4311744513,"GTP Round robin"),(5067190966567520076,4311744513,"FRG Single"),(5067190966566536528,4311744513,"FRG Dependent"),(5067190966566863948,4311744513,"FRG Idle"),(5067190966566341708,4311744513,"FRG Alloc New"),(5067190966567456324,4311744513,"FRG Round robin"),(4847084131849553740,4311744513,"CDM Single"),(4847084131848570192,4311744513,"CDM Dependent"),(4847084131848897612,4311744513,"CDM Idle"),(4847084131848375372,4311744513,"CDM Alloc New"),(4847084131849489988,4311744513,"CDM Round robin")),"IOReportChannelInfo"={"IOReportChannelUnit"=7205759403792793600},"IOReportSubGroupName"="vChannel Selection"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((5139821241828376576,4311744513,"Channel 0"),(5139821246123343872,4311744513,"Channel 1"),(5139821250418311168,43 [... cut ...] "="Spill Buffer Metadata Usage Histogram"},{"IOReportGroupName"="GPU Stats","IOReportChannels"=((6003392098212201796,137439215619,"Spill Buffer Unused Blocks per Task")),"IOReportChannelInfo"={"IOReportChannelConfig"=<01000000000000000000000020000000>,"IOReportChannelUnit"=0},"IOReportSubGroupName"="Spill Buffer Metadata Unused Blocks Per Task Histogram"})
      "IOMatchCategory" = "IOAccelerator"
      "IOProbeScore" = 10000
      "KDebugVersion" = 4294967296
      "SurfaceList" = ()
    }

```

exelban, 2022-09-23:
Hi. Thanks for the feedback.
What exactly do you mean by stats not showing?
Is the GPU missing in the GPU view or some values are not available in the details?

Mesmer88, 2022-09-28:
Sorry for the delay.
The GPU is not showing in the status bar

[image]

[image]

[image]

exelban, 2022-09-29:
For 99% it's a problem with a menu bar. And it's just not visible. Try to remove some elements from the menu bar, and it will appear.

closed 2022-09-29 by exelban