exelban/stats #1108: Bug report:

Jerry23011, 2022-10-03:
**Description of the bug**
Hi, I find that the bar chart on the menu bar is not showing the right icon. Just discovered this when I was checking the translations. Please see picture below.
[image]

**Details:**
 - Device: Macbook Pro 2021
 - macOS: 12.6
 - Application version: latest (2.7.37)

exelban, 2022-10-03:
Hi. What do you mean?
As I can see on the screenshot it shows the right widget.

Jerry23011, 2022-10-03:
So the bar chart shows one column only?
I thought there would be multiple columns as the icon in the Dashboard shows.

exelban, 2022-10-03:
If you want to see the usage per core you need to enable that option.

closed 2022-10-03 by exelban