exelban/stats #1110: Feature request: more CPU bar graph options for Apple Silicon

tycho, 2022-10-03:
Right now the bar graph options for CPU usage break down into two types: show a bar per-core, or show a single bar for all cores put together.

The real estate on my menu bar is rather limited thanks to the notch, but I don't want to lose too much detail in Stats. It seems like it would be nice to have a feature that allows condensing per-core bar charts into two bars: one for efficiency cores, one for performance cores. It looks like the values are already being computed for that, as the menu for it has two aggregate percentages for efficiency/performance cores.

exelban, 2022-10-04:
Hi. Thanks for your feedback.
Yeap, I think it could a nice feature. Will implement that.

tycho, 2022-10-05:
Actually now that I'm looking at it, I think the calculation for the percentage utilization for "efficiency cores" and "performance cores" metrics in the menu are not right. With my 2 efficiency cores (M1 Max) fully loaded by backupd doing a Time Machine backup, it only shows "18%" for efficiency cores. I'm guessing it's calculating the metric as "percentage of all logical CPUs" instead of "percentage of efficiency cores".

closed 2022-10-16 by exelban

tycho, 2022-10-16:
@exelban I think you missed the comment above, the usage percent calculation isn't right. If I start a task eating 4 full cores on an M2 (4P/4E), the bar for performance cores only goes up to ~50%.

tycho, 2022-10-16:
Filed bug #1120 to ensure this gets noticed. :)