exelban/stats #1111: ICMP not working

almottier, 2022-10-06:
**Describe the bug**
I am just trying the new ICMP feature. It stays red but I have an internet connection over Wifi. I tried forcing Wifi interface in the settings but I have the same result.

[image]

**Details:**
 - Device: MacBook Pro (14-inch, 2021)
 - macOS: 12.6 (21G115)
 - Application version: 2.7.37

exelban, 2022-10-06:
@luckman212 I need to add logging to the file. So I will post a new build. Thanks.

exelban, 2022-10-06:
ok, there is a debug build. It writes some logs related to the ICMP to the file `Documents/log.txt`. There must be no sensitive data inside this log. Please share the logs to better understand why it does not work properly.

[Stats.dmg.zip](https://github.com/exelban/stats/files/9726376/Stats.dmg.zip)

luckman212, 2022-10-06:
Ok, got it 👍 

## Video

on left is a continuous ping to `9.9.9.9` and on right is `tail -f log.txt` -- you can see Stats red dot in menubar...

https://user-images.githubusercontent.com/1992842/194359492-372dc544-e49e-4a8e-887b-c1c413186c25.mp4

## Log

[log.txt](https://github.com/exelban/stats/files/9726471/log.txt)

## Settings

[image]

[image]

exelban, 2022-10-06:
Hmm, very strange. Could you please try this build (it collects more information about why the signal is false)?
[Stats.dmg.zip](https://github.com/exelban/stats/files/9726536/Stats.dmg.zip)

exelban, 2022-10-06:
Yeap, exactly. The packets that are coming to the Stats most probably are from the terminal. In theory, I can disable fingerprint checks. But I'm not sure if it's ok. Because, for example, there could be some internal ICMP packets from the local network. And this check will lose any sense.

@almottier please check if you have the same situation.

luckman212, 2022-10-06:
I don't think fingerprint should be disabled, I haven't looked at the code because I have to leave now but have you taken a look at https://github.com/samiyr/SwiftyPing ? Maybe some ideas there that you can borrow from.

Because when I compiled that Test project, it works fine pinging at the same time as Terminal...

[image]

almottier, 2022-10-07:
> @almottier please check if you have the same situation.

I confirm I have the same situation. As soon as I ping from a terminal, I start to have a red dot in Stats.

exelban, 2022-10-09:
Thanks, I will try to fix that.

luckman212, 2022-10-14:
I compiled a build based on the latest `master` branch, and confirm that the fix is working @exelban

luckman212, 2022-10-15:
Anyone else notice that this is issue #`1111` and the default ICMP target is also `1.1.1.1`... cosmic irony? 👻

closed 2022-10-16 by exelban