exelban/stats #1114: Network Doesn't Properly Report Traffic On IPv6 Only Networks

Rishi556, 2022-10-09:
**Describe the bug**
When I'm on an IPv6 only network(no ipv4 address at all) the network values stop working. The numbers displayed on the menu bar and the graph is 0 even though when I look at top processes, I can see Chrome is actively using network(it was downloading a file).

Recreating the bug is simple. Connect to a network that supports dual stack. Disable IPv4 on your device by going to System Preferences, Network, Select the network interface you use, Advanced, TCP/IP, Configure IPv4 to off. Then go to a browser that supports and try and browse the IPv6 powered internet(https://se1.files.privex.io/images/iso/ubuntu/22.04.1/ this will work, I was pulling the [this](https://se1.files.privex.io/images/iso/ubuntu/22.04.1/ubuntu-22.04.1-live-server-amd64.iso) file), you will notice that the bar only displays 0.

[image]

**Details:**
 - Device: MacBook Pro (14-inch, 2021)
 - macOS: 12.5.1
 - Application version: 2.7.37

Rishi556, 2022-10-09:
After further review, it looks like it does work when set to process based rather than interface based for reader type. Would it be better to default to that rather than interface based?

exelban, 2022-10-09:
The problem is not in the app, but in the interface or interface driver.

closed 2022-10-09 by exelban