exelban/stats #1118: [Feature request] Add option to show CPU usage as a "sum of all cores" instead of "average"

sullerandras, 2022-10-15:
At the moment the "CPU percentage" seems to be the average of all CPU cores, and it goes from 0% to 100%, so if all cpu cores are busy working then it will show 100% usage.
I'd prefer to see the cpu percentage as a number between 0% and 800% if the computer has 8 cpu cores. So it'd simply sum up the cpu usage of all the individual processes.

Probably it'd also be useful have the option to sum up the cpu usage per cpu type related to https://github.com/exelban/stats/issues/1110

exelban, 2022-10-25:
Hi. Thanks for the feedback.
And sorry for such a long response. All widgets and views are set up to visualize the data in the range between 0 and 100.
Adding this feature will require tweaking all widgets, charts, and views. I'm not ready to do that, and for sure will have no time for that.

closed 2022-10-25 by exelban

sullerandras, 2022-11-15:
Thank you for making this project open source! I started working on updating the code: https://github.com/sullerandras/stats/pull/1
So far it's a one line change, with the absolute minimum that i needed.