exelban/stats #1120: efficiency/performance core usage percentage incorrect

tycho, 2022-10-16:
**Describe the bug**
The efficiency/performance core utilization percentage is incorrectly calculated in both the dropdown menu and in the new menubar cluster grouping feature.

For example:

- If I run a 4-thread workload that eats up all the performance cores on my M2 (4P/4E), then the calculation for performance cores is 50% instead of 100%.
- If I run a time machine backup on my M1 Max (8P/2E), the usage calculation for efficiency cores gets to 20% instead of 100%.

I think what's happening is that it's calculating the efficiency/performance usage percentage as "percent of all cores" rather than percent of that cluster type.

**Details:**
 - Device: MacBook Pro 14" (M1 Max), and MacBook Air (M2)
 - macOS: Ventura
 - Application version: 2.7.39

tycho, 2022-10-16:
Oh, also if I fully load (8-thread workload) an Apple M2, both performance and efficiency percentages are at 50, which obviously doesn't make sense.
[image]

tycho, 2022-10-16:
I think this is probably the solution (tested locally):
```diff
diff --git a/Modules/CPU/readers.swift b/Modules/CPU/readers.swift
index 40496be..6dc319c 100644
--- a/Modules/CPU/readers.swift
+++ b/Modules/CPU/readers.swift
@@ -143,8 +143,8 @@ internal class LoadReader: Reader<CPU_Load> {
                 return 0
             }

-            self.response.usageECores = eCoresList.reduce(0, +)/Double(cores.count)
-            self.response.usagePCores = pCoresList.reduce(0, +)/Double(cores.count)
+            self.response.usageECores = eCoresList.reduce(0, +)/Double(eCoresList.count)
+            self.response.usagePCores = pCoresList.reduce(0, +)/Double(pCoresList.count)
         }

         self.callback(self.response)
```

tycho, 2022-10-17:
Also, the "cluster grouping" option on an Intel mac seems to just result in two empty bars in the menubar, probably should hide that option on non-ARM64?

exelban, 2022-10-17:
Hi. Thanks for the feedback.
And thanks for the solution. I will take a look at this.

closed 2022-10-23 by exelban