exelban/stats #1122: [Feature request] Export Settings

mjr2595, 2022-10-16:
It would be super awesome if we had the option to export our settings. I like to play around with theming and being able to quickly export/import settings is very helpful. Also helpful when setting up a new computer.

...Unless I'm just not seeing this option and it already exists.

mjr2595, 2022-10-16:
commenting just to say thanks for all your hard work. Love this application!

exelban, 2022-10-16:
Duplicate of #1093

closed 2022-10-16 by exelban

mjr2595, 2022-10-16:
Ah thanks for pointing out the previous request. For anyone else seeing this, here's a [link](https://github.com/exelban/stats/issues/801#issuecomment-1020517746) to a previous request on how you can find/export the settings.