exelban/stats #1123: Strange spacing

Catalin9090, 2022-10-16:
When only this function is enabled I get some spacing after the power meter. 
[image]
[image]

exelban, 2022-10-17:
Hi. Thanks for the feedback.
What's happening if you click on the empty space?

Catalin9090, 2022-10-17:
Nothing.

exelban, 2022-10-17:
no popup does not appears?

Catalin9090, 2022-10-17:
No popup. But if i enable and disable the cpu or memory then the power number gets more to the left, leaving a larger space.

exelban, 2022-10-17:
So please disable the modules one by one, and take a look at what module will hide the empty space.
Looks like it's a widget that does not have any data to show.

Catalin9090, 2022-10-17:
This is exactly what I did. None of them are on. Restarted the mac and the space is still present. Mac Studio M1 Max.

exelban, 2022-10-17:
Em, if non of the modules do not dissappear the empty space. So maybe not Stats reserve this space?

Catalin9090, 2022-10-17:
Don’t know. First time using the power widget. Normally I was using the cpu, gpu and memory. All are fine with no strange spacing.

exelban, 2022-10-17:
Please disable the `Sensors` module. And check if empty space is dissappear.

Catalin9090, 2022-10-17:
It does not
[image]
The app running now is tgpro.

Catalin9090, 2022-10-17:
the space is gone
[image]

Catalin9090, 2022-10-17:
[image]

exelban, 2022-10-17:
Ok, please disable all modules.

Catalin9090, 2022-10-17:
I did and there is the same behaviour

Catalin9090, 2022-10-18:
Yes

exelban, 2022-10-18:
emm, it's not possible.

Catalin9090, 2022-10-18:
I’ve send you all the pictures you requested and now you tell me it is impossible.

exelban, 2022-10-18:
Please check if you have no enabled `Merge widgets` option in any module.

Catalin9090, 2022-10-18:
yep. that fixed the issue. But It required a app restart as when I disable merge widgets the disabled widget would be shown next to the power widget. Thank you

exelban, 2022-10-18:
Thanks for reporting and spending your time to detect the problem. I will try to fix it.

Catalin9090, 2022-10-18:
Thank you! Amazing job and great patient.

closed 2022-10-23 by exelban