exelban/stats #1125: [Feature Request] Stat logging and data export

jf-wm, 2022-10-18:
Currently the program only shows the current stats, but does not offer an option to export them regularly (e.g. every minute) into a external document (e.g. csv). Especially for the battery it would be really helpful to log and export the stats in order to analyse the behaviour during a single or multiple charge / discharge cycles.

exelban, 2022-10-18:
Hi. Unfortunately, it cannot be done like this. Because it's not just simply writing to the file till memory will end. It will require a lot of logic (what exactly db have).
There is only one way how it could be done is some time series db. But it requires having a db on the local machine. So it cannot be implemented like this. Maybe it could be done via server with tsdb. But I'm not sure that will add this functionality. If you need to monitor specific stats there are nice python tools that allow you to do that.

closed 2022-10-18 by exelban

iWhacko, 2025-12-05:
As crashes can sometimes occur due to overheating, logging would be great. Wouldn't a simple sqlite db be perfect to log data like this?