exelban/stats #1126: Focus is not returned back to previously active window

dimitrik-fr, 2022-10-21:
**Describe the bug**
When you click on the menu-bar Stats icon to get a more detailed info about any given metrics, the focus is not returned back to your previously active window (opposite to all other in-menu apps) -- this is requiring each time you click again on your window to bring the focus back and make it active again. 

Sure, it's a minor thing (comparing to overall great stuff the apps is providing), but over a time becomes pretty annoying ;-))

**Details:**
 - Device: Macbook Pro 2020
 - macOS: 12.6
 - Application version: 2.7.39

exelban, 2022-10-23:
Hi. Thanks for the feedback.
It's because the popup view it's not a normal window. And it cannot be normally switched by tab for example.

closed 2022-10-23 by exelban

dimitrik-fr, 2022-10-23:
I'd agree with you if it was the same case for all menu-bar apps, but it's not so ;-))
look on BitBar for ex. which is doing similar things as you in menu bar -- https://macmenubar.com/bitbar/ -- once popup is closed, the focus is always back to the previously active window, np.