exelban/stats #1128: Download unsuccessful; Finder displays

SyntaxAdl, 2022-10-24:
Clicking on "Code" ---> "Download ZIP" downloads a folder, not executable file format. Application contents show as files and folders within main "stats-master" folder when clicked upon. 

 - Device: iMac 24.5" (2021) 
 - macOS: Monterey 12.6
 - Browser version: Safari 16.0 
 - Application version: Stats v2.7.40

[image]

[image]

I'm new to GitHub so please list possible configurations to change this if necessary as a solution - i.e. this is supposed to happen by default.

BitesPotatoBacks, 2022-10-24:
@SyntaxAdl You would want to download the Stats.dmg file from the [latest release](https://github.com/exelban/stats/releases/tag/v2.7.40)

exelban, 2022-10-24:
Hi. It's because you downloading the source code, not the compiled application.
You can download the application [here](https://github.com/exelban/stats/releases/tag/v2.7.40).

closed 2022-10-24 by exelban