exelban/stats #1129: Odd network activity when Network (ICMP) module is enabled

luckman212, 2022-10-24:
**Describe the bug**

I don't understand yet _why_ this is happening, but when I have the Network module enabled, with ICMP host pinging, I am seeing a flood of `mDNSResponder` activity in my console (repeats every second). Various different IPs are being queried from all over the place. Not sure if this relates to the syscalls to `curl api.serhiy.io` etc (not sure how that could be the case since there are only 2 IP addresses defined there...)

**Details:**
 - Device: Mac Mini M1
 - macOS: 12.6
 - Application version: 2.7.40

**Screenshot**

[image]

If you prefer a Terminal view, this can be done with e.g. (requires jq)
```shell
log stream --predicate 'subsystem contains "com.apple.mDNSResponder"' --style ndjson | tail -n+2 | jq -r 'select(.eventMessage|test("1.0.0.127.in-addr.arpa")|not) | [.timestamp,.eventMessage] | @tsv'
```
As soon as I switch off the Network module, the activity stops. 🤷

exelban, 2022-10-25:
Hi.
Thanks for the feedback. I cannot reproduce that. Not on apple silicon machine, not on intel. There is no these IP querying. You can disable ICMP by leaving the address empty. But ICMP if you are using an IP address does not use a DNS resolver. Because it already has an IP. Also, it cannot be IP fetching because it runs max 1 per hour (if you change these settings). By default, it does not refresh at all.

luckman212, 2022-10-25:
@exelban Thanks for looking. It is so crazy, I agree it should not be possible but somehow- it is. I tested & was able to reproduce every time, as soon as I switch off the Network module, the queries stopped. I can send a screen recording.

I will try compiling from source and see if it persists. I am on Ventura 13.0 now as well.

exelban, 2022-10-25:
Oh, I have no info about Ventura compatibility right now. So at least for the next few releases, there will be some fixes for it. Will be back when will have more info.

exelban, 2022-10-25:
Ok, I have installed the Ventura. And still cannot reproduce that.

exelban, 2022-10-25:
@luckman212 in the logs there is a PID, please check what process has this PID.

luckman212, 2022-10-26:
The PID keeps changing as far as I can tell @exelban 

I made a video. I don't know how to explain this behavior.  I also tried building from source, and the app produced the same behavior. Makes no sense!

https://cln.sh/dS90d1

luckman212, 2022-10-26:
@exelban I am pretty sure the source of the problem is in the `resolve()` function.

https://github.com/exelban/stats/blob/65c15a7b6fa36f9e80c879723aea99c903680dbe/Modules/Net/readers.swift#L785

I think `CFHostStartInfoResolution` may return `false` or `nil` for pure IP addresses

https://github.com/exelban/stats/blob/65c15a7b6fa36f9e80c879723aea99c903680dbe/Modules/Net/readers.swift#L789-L794

so `self.lasthost` never gets assigned a value and thus gets triggered by on every loop of `read()`:

https://github.com/exelban/stats/blob/65c15a7b6fa36f9e80c879723aea99c903680dbe/Modules/Net/readers.swift#L656-L660

exelban, 2022-10-26:
thanks, will take a look

exelban, 2022-10-27:
@luckman212 hi. And does it stops resolving DNS if you disable ICMP check?

exelban, 2022-10-27:
https://github.com/exelban/stats/blob/58cb2557805fbd23699c7f5c47461c39f80b38b3/Modules/Net/readers.swift#L785-L786

It does not matter if `resolve()` returns nil. Because `lastHost` is set at the beginning of the function. So every time `resolve` is called, the last host is updated. Even if it does not resolve the host.

luckman212, 2022-10-27:
@exelban Hmm. No, the queries did not stop when I cleared the ICMP host field. They 100% do stop when I turn off the Network module entirely though. That is still a mystery.

exelban, 2022-10-27:
So it's not related to the ICMP. Could you please run that build and check the logs?

[Stats.dmg.zip](https://github.com/exelban/stats/files/9881905/Stats.dmg.zip)

luckman212, 2022-10-27:
Sure - I ran it but, is it supposed to output an extra logfile? I don't see one.

exelban, 2022-10-27:
Nope, I mean in the console. mDNS logs

luckman212, 2022-10-27:
Ok, that's what I thought. The debug build causes the same kind of flooding mDNS output in the console as soon as Network module is enabled.

exelban, 2022-10-27:
ok, so it's not an IP receiving. Try this one.

[Stats.dmg.zip](https://github.com/exelban/stats/files/9882554/Stats.dmg.zip)

luckman212, 2022-10-28:
Hi @exelban - same results.

[image]

exelban, 2022-10-28:
@luckman212 please change the reader type to `Interface based` and check if something is changed.

exelban, 2022-10-28:
It's not a problem with Stats. But with nettop. These all builds were with disabling some modules (it's why connectivity does not work). I cannot do anything with nettop flooding.

closed 2022-10-28 by exelban

luckman212, 2022-10-28:
Ok @exelban — thank you for looking into this. But, before you close...

How about adding `-n` flag to `nettop` [here][1] and [here][2]

https://github.com/exelban/stats/blob/58cb2557805fbd23699c7f5c47461c39f80b38b3/Modules/Net/readers.swift#L244

I tested it and it solves it also on my system. We don't need name resolution to calculate bandwidth usage anyway?

[1]: https://github.com/exelban/stats/blob/58cb2557805fbd23699c7f5c47461c39f80b38b3/Modules/Net/readers.swift#L244
[2]: https://github.com/exelban/stats/blob/58cb2557805fbd23699c7f5c47461c39f80b38b3/Modules/Net/readers.swift#L442