exelban/stats #1130: Thermal sensors not working as expected on M2 Macbook Air

jacobdman, 2022-10-24:
**Describe the bug**
A clear and concise description of what the bug is.

On computer startup the thermal sensor stat is missing, even though I have it turned on for CPU and GPU averages. The sensor shows after turning one of them off (in my screenshot I have turned it off and back on), but now only shows a "no" icon. When I click on the icon, I can see the accurate temps.

[image]
[image]
[image]

**Details:**
 - Device: MacBook Air (M2, 20222)
 - macOS: macOS Monterey, Version 12.5
 - Application version: 2.7.40
[image]
[image]

exelban, 2022-10-25:
enable the module)

closed 2022-10-25 by exelban

jacobdman, 2022-10-25:
Wow, that was dumb. Thank you.