exelban/stats #1138: [Feature Request] Set cpu max temp

tckackley, 2022-10-27:
Hello @exelban! Thank you for sharing this great app with us.

It would be great if we could have the ability to set a max cpu temp and then the fan would automatically adjust to keep the cpu temp down. Something like Macs Fan Control. https://crystalidea.com/macs-fan-control

I understand this could potentially require a great deal of work so I understand if this is not something on the top of the priority list.

exelban, 2022-11-05:
Hi. Thanks for the feedback.
Sorry for the late reply. I needed to think about that. And I think it could be implemented with time. It requires changes in the UI, but it could be done.

Thomas-Cee, 2024-04-09:
What an incredible app. Wow, thank you for the incredible work. 
I use the built-in Stats manual fan control to keep CPU temps a little lower than the computer does stock.
This feature would be a neat addition; a little more fan control wither,
- Up the fan percentage
- Add a CPU top target temp that could be lowered
- Higher fan speed priority 
I feel like any tweak like that would be amazing. 

For daily use, I keep the fan manually a little higher than standard, and I raise it a little more when doing intensive things with the computer. I do find myself fiddling with it alot though.

Egregius, 2024-04-12:
Good idea! Just started using Stats to replace MenuMeters, seems to be abandoned. Stats looks really good, like it. 
Just the fan control could be better with thresholds or optimally templates. Until then I'll keep using Macs FanControl for that:
[image]
I do like those templates. The 45-100 is my default. Keeps the fan with normal usage at around 1300rpm. As soon as a process is using 100cpu I hear the fans spinning up and I can check whats causing it.

exelban, 2024-05-05:
Hi. After some consideration, I'm going to close this issue because it will not be implemented.

There are a few reasons for that:
- first of all, I'm more interested in having a monitoring tool rather than an app for controlling the Mac. I agree to add fan control because it could be useful for some users (I have never used it since adding it to the app)
- on the newest osx or Macs (don't know the exact reason) Apple blocks the external control. That means shortly this feature will be not impossible at all because there will be no way to control the fan.
- it requires a lot of work to make this feature and there are a lot more features that more critical from my point of view

If you want to control the fans I really recommend using specified tools that are specialized in that.

closed 2024-05-05 by exelban