exelban/stats #1142: App permission issue

ngshijun, 2022-10-30:
**Describe the bug**
I've enable accesibility settings for **stats** but everytime when I try to adjust the fan rpm, the system will still prompt me to enter my password.

**Details:**
 - Device: MBP 2021
 - macOS: 13.0
 - Application version: 2.7.41

exelban, 2022-10-30:
Yes. Don't use it.

closed 2022-10-30 by exelban

ngshijun, 2022-11-03:
Hi, but the issue doesn't happen before I update to macOS 13. Is this a macOS issue?

exelban, 2022-11-03:
It always was like this. There is no option to control the fans without root access in macOS.

ngshijun, 2022-11-03:
Thanks for the reply. Just to confirm, even if accessibility settings is enable for the app, we still need to key in password each time we want to make changes to the fan speed?

exelban, 2022-11-03:
Yes. I'm not sure if Stats needs the accessibility settings enabled. Yes, every restart will require the password because all features don't require the root. So normally the Stats does not run with the root.

ngshijun, 2022-11-03:
Noted, thank you very much.