exelban/stats #1145: The memory information page cannot view high-occupancy processes

ChangCXQ, 2022-11-02:
Memory footprint info, I don't think it's the correct interface, but no matter what I do, it just shows the info as shown in the screenshot.
[image]

exelban, 2022-11-02:
Hi. Thanks for the feedback.
Please share the output of the `top -l 1 -o mem -n 8 -stats pid,command,mem`.

ChangCXQ, 2022-11-02:
Apple M1，macOS 13.0 (22A380)
Stats 2.7.41
[image]

exelban, 2022-11-02:
Hmm, very strange. Please try to change the top process number. For example 8 -> 0 -> 8. It will restart the top reader.

exelban, 2022-11-02:
Also, please check if processes are visible in the CPU module.

ChangCXQ, 2022-11-02:
> Also, please check if processes are visible in the CPU module.
Process information in the CPU module:
[image]

exelban, 2022-11-02:
Only I can do it creates a debug build that will collect the logs to the file. Do you want to help with that?

ChangCXQ, 2022-11-02:
> Only I can do it creates a debug build that will collect the logs to the file. Do you want to help with that?

Yes, I would love to do that. Because I think Stats is an excellent program and I enjoy using it.

exelban, 2022-11-02:
Thanks.
Please run that build and open the RAM popup for a few seconds. After that, you will find a file `log.txt` in the `Document` folder. Please share it. It allows figure out why there are no data.

[Stats.dmg.zip](https://github.com/exelban/stats/files/9922499/Stats.dmg.zip)

ChangCXQ, 2022-11-02:
> Thanks. Please run that build and open the RAM popup for a few seconds. After that, you will find a file `log.txt` in the `Document` folder. Please share it. It allows figure out why there are no data.
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/9922499/Stats.dmg.zip)

[log.txt](https://github.com/exelban/stats/files/9922530/log.txt)
I had the RAM window open during that time period at
November 3, 2022, 00:47 CST

exelban, 2022-11-02:
Thanks. I see the bug.

Please try this build: 
[Stats.dmg.zip](https://github.com/exelban/stats/files/9922657/Stats.dmg.zip)

ChangCXQ, 2022-11-02:
> Thanks. I see the bug.
> 
> Please try this build: [Stats.dmg.zip](https://github.com/exelban/stats/files/9922657/Stats.dmg.zip)

[image]
Very good, the bug is fixed.

exelban, 2022-11-02:
It's not fixed yet. I will close the issue when the release with that fix will be available. To prevent duplicating the issue.
Thanks for your help)

ChangCXQ, 2022-11-02:
> It's not fixed yet. I will close the issue when the release with that fix will be available. To prevent duplicating the issue. Thanks for your help)

Okay, okay, I'm waiting for the new version to be released.

closed 2022-11-06 by exelban