exelban/stats #1147: Colorize option for battery does not color battery at 100%

skylon07, 2022-11-05:
Firstly, thank you for making this app. It is very well designed and maintained, and my experience so far has been very smooth! I love being able to see my computer's vitals in the menu bar. It makes it so much easier than pulling up Activity Monitor every time something goes wrong.

**Describe the bug**
I noticed one small issue when using the "Colorize" option for the battery widget. When my battery is fully charged, the battery color is still white. The moment it hits 99% though, it turns green. Below is an example of the "Colorize" option being activated, and yet the battery is still white. 

[image]

This is totally a nitpick, and it's a little funny that it bothers me so much. Anyway, aside from this, I haven't seen any other issues with the app. Mark that a victory, because I do software testing for a living! Thanks again for the very well made app.

**Details:**
 - Device: Macbook Pro M1
 - macOS: Monterey 12.6
 - Application version: 2.7.41

exelban, 2022-11-06:
Hi. Thanks for the feedback.
The battery is colorized only when the battery level is different than 100%.

closed 2022-11-06 by exelban

skylon07, 2022-11-06:
Oh, I didn't realize this was intentional. Is there a reason for this, or is it just a design choice?

exelban, 2022-11-06:
Inform the battery is full.

skylon07, 2022-11-06:
Ah, I see. That makes sense, thinking about cases where the user might not be displaying the percentage. Thank you for explaining!