exelban/stats #1149: Bluetooth doesn't work

iamredd, 2022-11-06:
**Describe the bug**
First, I've multiple device connect to my macbook, mouse, keyboard, touchboard but it only show one, please see the attachment.

Second, bluetooth  icon didn't reveal 

**Details:**
 - Device: MacBook Pro 16-inch 2021
 - macOS: Ventura 13.0
[image]

 - Application version: 2.7.41

exelban, 2022-11-06:
It means that these devices battery levels do not available via API.

closed 2022-11-06 by exelban

iamredd, 2022-11-06:
How could I solve this issue?

exelban, 2022-11-06:
Maybe contact the manufacturer and ask to use the same way to report the battery level as apple devices do that (and many more).

exelban, 2022-11-06:
Or maybe you know the other way to receive the battery level from the Bluetooth device. I have implemented 2 and looks like that device does not support any of them.

iamredd, 2022-11-06:
Thanks! But I'm not very familiar with coding, so maybe I need to find something else.