exelban/stats #1150: Unable to change fan speed on macOS Ventura

RaazKH, 2022-11-06:
**Describe the bug**
I upgraded to `macOS Ventura` today and I have been unable to change my fan speed. If I try setting it to manual/automatic it will ask for administrator credentials.
[image]
After entering the credentials nothing happens.

I upgraded my stats to 2.7.42 at the same time so I assumed it wanted admin credentials to give permission for the application to make changes to fan speed but that was not the case.

**Details:**
 - Device: Mac mini 1018
 - macOS: Ventura 13.0
 - Application version: 2.7.42

exelban, 2022-11-06:
And what is the problem here?

RaazKH, 2022-11-06:
The fan speed does not change. If i try to increase the RPM I get asked for admin credentials and nothing happens.
1. I try to adjust the fan speed (lets say clicking on the snowflake icon)
2. I am asked fir admin credentials (see above screenshot)
3. I enter administrator credentials
4. I am asked for admin credentials a second time
5. I enter them a second time
6. The stats menu closes and the fan speed does not change.
7. If I re-open the menu I see this:
[image]
It says 1690 for the fan but on the max setting it should be around 4000.
I can tell the fan is not changing. Do you think something was changed in macOS Ventura regarding permissions for applications to change system settings?

exelban, 2022-11-06:
There was no change in the new macOS regarding the SMC. I recheck and Stats can control the fans on the macOS Ventura. And what if you switch to the manual mode and set the speed at 50%?

RaazKH, 2022-11-06:
So after clicking manual I first have to enter admin credentials after which the menu closes
Then I open the menu again and shows manual as selected
If I drag the slider to 50% it will close the menu once I let go of my mouse button and another popup comes asking for admin credentials. Entering the credentials doesn't seem to change anything and if I open the menu again I see this:
[image]

exelban, 2022-11-06:
Looks like you have some problems with the permissions. And the application cannot make changes in the SMC.

RaazKH, 2022-11-06:
How do I give the app permissions to make changes in the SMC? There is no SMC option under Privacy & Security which would allow me to grant permissions to stats.app

exelban, 2022-11-06:
SMC - it's a System Management controller. There is no policy for that. It's an integrated part of mac.
You can try to change the fan speed with that `/Applications/Stats.app/Contents/Resources/smc fan -id 0 -v 3000`. 
You can see all commands with the `help` arg.

**Important notice, manipulating with SMC it's not 100% safe. And there is no warranty that mac will work fine when you do that.**

RaazKH, 2022-11-06:
I get this message when I run the command as a normal user: Error write: (iokit/common) privilege violation
I ran the same command using `sudo` and there was no error message but the fan speed did not change.

RaazKH, 2022-11-06:
I can still get read from the SMC though, here is what I get when I run `/Applications/Stats.app/Contents/Resources/smc fans`:
[image]

exelban, 2022-11-06:
Do you have an administrator account on mac?

RaazKH, 2022-11-06:
Yes.
The other account(not administrator) is the one I use most of the time.

RaazKH, 2022-11-08:
Ok, thanks for looking into it 👍

RaazKH, 2022-11-11:
Hey I was trying to fix it myself because I really like this application and nobody else has an application as good and I noticed this issue: https://github.com/exelban/stats/issues/430 and the guy in this comment has the same exact issue as me https://github.com/exelban/stats/issues/430#issuecomment-864183006

And my Mac also has a T2 chip. I also downloaded [this](https://crystalidea.com/macs-fan-control/download) and I was able to change my fan speed just fine.

Is there a way for me to completely erase all data associated with this application? Maybe one of the configuration files got corrupted...

exelban, 2022-11-11:
Yeap, and that was fixed.