exelban/stats #1151: Feature Request: add a pin icon to make popover window stay visible

muescha, 2022-11-06:
## what i see:

- if i click on a graph in menubar then a popup window is shown
- i assumed the left icon is a "stay forever" (but it seems this icon has no function?)
[image]
- i assumed when i click on the gear icon that i have a context menu to pin the window, but this icon only opens the settings

## what i expect:

- a pin icon where i can click to that i can pin the window

## workaround:

- later i discovered that i can drag from the title bar of the popup window to get it deattached and stay forever 
- but i think this is not intuitive for new users

exelban, 2022-11-06:
You can do it by dragging the popup window.

closed 2022-11-06 by exelban

muescha, 2022-11-06:
now after many tries i found it accidental - now i know it is can done by this way = but this not intuitive

adding a move handle icon would be an enhancement `arrow.up.and.down.and.arrow.left.and.right` (SF Symbol):
[image]