exelban/stats #1153: [Feature Request] Show RAM swap usage in widget

LinyiWu, 2022-11-07:
Thank you for this awesome app!

It would be great if there is a RAM Swap Widget to show swap usage.

exelban, 2022-11-07:
Hi.
I don't see any reason to have a widget for a swap.

closed 2022-11-07 by exelban

rizrmd, 2022-12-31:
Monitoring swap is great tool to maintain your SSD health, see: https://www.youtube.com/watch?v=MZuv4TIjk-I

jvanakker, 2024-04-09:
@exelban Why is there no reason to display swap usage? For anyone with an SSD mac, having the ability to show SWAP usage would be one of the most useful tools of this whole app. Having ***any*** amount swap usage will significantly contribute to the limited TWB (lifespan) of your SSD.

Since SSD's are soldered in, there is not much you can do, but **swap** usage is one of the few things that you actually have control of, to optimize the lifespan of the SSD.

Two requests:

- The current Stats version (2.10.7) lets me set a warning at 0.5 GB of swap, but can this be changed so we get a notification with ANY swap usage? (even 250 MB would significantly contribute to the limited lifespan).
- The possibility to show the actual swap value in the widget would be great. Currently you have to click on RAM widget, then see the swap there.

skylon07, 2024-10-14:
Just curious, is there any reason why the current RAM widget doesn't work for you guys? It shows in the "Details" section "App", "Wired", "Compressed", "Free", and finally "Swap".

jvanakker, 2024-10-15:
> Just curious, is there any reason why the current RAM widget doesn't work for you guys? It shows in the "Details" section "App", "Wired", "Compressed", "Free", and finally "Swap".

You mean the popup view that shows up when you press the menu bar widget? So that's only _after_ you press the RAM widget. You can't configure it show the swap usage number itself _**in**_ the widget. Just checked it with latest v2.11.14.

Also, I noticed the swap notification function still has no hysteresis/damping. So once you the set swap notifications on, and the threshold is reached, you get a seemingly endless number of notification popups to click away.

I did notice that with the latest version you can set the swap number to any amount (previously limited to 0.5 GB, so keep up the good work!

skylon07, 2024-10-16:
Ah, I see what you mean now. Thanks for clarifying