exelban/stats #1155: [Feature request] Add ability to switch between integrated and discrete GPUs

Fail-Safe, 2022-11-08:
Humbly submitting this request to see if you might entertain adding a toggle to force-switch between integrated and discrete GPUs on systems where both are available. I was a long-time user of [gfxCardStatus](https://gfx.io) and removed it in favor of Stats. But, I do miss the ability to force my system onto the discrete GPU when I desire that.

Ref: https://github.com/codykrieger/gfxCardStatus/blob/master/Classes/GSMenuController.m#L154

Thanks!

exelban, 2022-11-08:
Hi. Thanks for the feedback.
There is somewhere in the issues answer on that feature. TL.RD: nope, because Stats is about monitoring and not being one tool for all. Better use the tool that specialized in that. Also, there is no discrete GPU in the MacBooks for a long time.

closed 2022-11-08 by exelban

Fail-Safe, 2022-11-08:
Thanks for following-up and letting me know the stance of Stats on this request.

For the sake of clarity, I do want to address your statement about discrete GPUs in MacBooks. I have a 2019 MacBook Pro (16") and definitely have two GPUs.

> Also, there is no discrete GPU in the MacBooks for a long time.

[image]

exelban, 2022-11-08:
Yeap, but in the new macs (apple silicon-based) there is no discrete GPU.