exelban/stats #1162: An empty and movable spacing appears when Stats is launched

Jerry23011, 2022-11-11:
**Describe the bug**
When I launch Stats, I find a movable space in the menubar with no content. When I hold command and drag it, it can take up space between other icons. I'm not sure if it's a bug or designed as intended.
The only menubar Widget I enabled is the Sensor Widget.
Please see the screen recording below

https://user-images.githubusercontent.com/89069957/201355917-29e4ca2b-96e0-4555-aaa5-d2b941de183b.mov

**Details:**
 - Device: Macbook Pro 14'
 - macOS: 12.6.1
 - Application version: 2.7.42

exelban, 2022-11-11:
Hi. Pretty sure that you have enabled another module that does not show any information, or there is enabled another widget without data. Please recheck if all other modules are disabled.

Jerry23011, 2022-11-12:
Hi, I closed all the module but it's still there, so I uninstalled Stats and reinstalled it, problem fixed, thanks.

closed 2022-11-12 by Jerry23011