exelban/stats #1166: Unable to set fan speed

Moulick, 2022-11-13:
**Describe the bug**
On setting the fan speed in the CPU widget, Stats asks for admin permissions 6 times. It's still is not able to change the fan speed

**Details:**
 - Device: Macbook Pro 14 Inch 2021 M1
 - macOS: Ventura 13.1
 - Application version: 2.7.42

exelban, 2022-11-13:
Hi. There is no option to change the fan speed throw the CPU module. Maybe you mean the sensors module?
There is no public release of macOS 13.1. So I have no idea why it does not work.

closed 2022-11-13 by exelban

tarushi-01, 2022-11-13:
> Hi. There is no option to change the fan speed throw the CPU module. Maybe you mean the sensors module? There is no public release of macOS 13.1. So I have no idea why it does not work.

Ah yeah, I mean't the sensor module. I'll re-verify once 13.1 is generally available.

Moulick, 2022-11-13:
> Hi. There is no option to change the fan speed throw the CPU module. Maybe you mean the sensors module? There is no public release of macOS 13.1. So I have no idea why it does not work.

Ah yeah, I meant the sensor module. I'll re-verify once 13.1 is generally available.