exelban/stats #1168: MenuItem issues, deselects and doesn't disable

johnalanwoods, 2022-11-13:
**Describe the bug**
When selecting a MenuItem (CPU, GPU, RAM, Sensor) etc, the menuItem immediately deselects. This is no coherent with other MenuBar Items, however, disregarding this, it leads to the ability to multiple selection like this:

[image]

**Details:**
 - Device: M1 Pro MacBook Pro, 32GB, 
 - macOS: 12.6.1
 - Application version: 2.7.42

**Expected behaviour**

MenuItem should stay selected when engaged, and deselect when another item is selected.

[image]

exelban, 2022-11-13:
It's because Stats uses a nswindow, and other items use nsmenu. And opening nsmenu doesn't trigger the focus as a result Stats window is not closing (the focus is not out from the Stats app).

closed 2022-11-13 by exelban

johnalanwoods, 2022-11-14:
OK Thanks