exelban/stats #1169: Sensors icon missing

zzjin, 2022-11-14:
**Describe the bug**
A clear and concise description of what the bug is.

**Details:**
 - Device: MBP14 m1 pro
 - macOS: 13.1.0
 - Application version: 2.7.43

The sensors icon is missing:
[image]

[image]

[image]

exelban, 2022-11-14:
enable the sensors to show

closed 2022-11-14 by exelban

zzjin, 2022-11-14:
What I Mean is the sensor icon show "blocked", but previous version it shows the right icon with sensor temperature.

[image]