exelban/stats #1172: Latest update broke sensors popup/widget

valynor, 2022-11-14:
mac Studio @ macOS 12.6.1

I have "power dc in" and "hottest cpu" enabled for the sensors widget. 
Since the update to 2.7.43 only "power dc in" is showing in the widget and the average+hottest cpu AND gpu sensors are missing from the popup completely. 

To fix this I have to go to the sensor settings and enable "show unknown sensors" -> the missing items show up immediately. I can even disable "show unknown sensors" after that and they keep showing. 

Unfortunately this happens on every restart of your app.

exelban, 2022-11-14:
Hi. Yeap, unfortunately, it's broken. There is already a fix for that. I will make a pre-release build with that. You can use it if you want.

exelban, 2022-11-14:
[Stats.dmg.zip](https://github.com/exelban/stats/files/10005566/Stats.dmg.zip)

LostInCompilation, 2022-11-14:
I'm not sure, if this is related: When I enable "Show unknown sensor", activate "Hottest CPU" then disable "Show unknown sensors" again, Stats crashed.
I cannot test the prerelease right now, since it will break my package versioning.

Crash log:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [14772]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.7.43 (418)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-11-14 21:26:26.8042 +0100
OS Version:            macOS 13.0.1 (22A400)
Report Version:        12
Anonymous UUID:        A2FA5528-1B49-8DC0-7BC2-772FFB6D12EB

Time Awake Since Boot: 89000 seconds

System Integrity Protection: enabled

Crashed Thread:        4  Dispatch queue: eu.exelban.Stats

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x00000001049f52b4

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [14772]

Thread 0::  Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	       0x1924c5834 objc_msgSend + 52
1   AppKit                        	       0x196119278 -[_NSTrackingAreaAKManager _invalidateCursorRectsForViewsWithNoTrackingAreas:] + 192
2   AppKit                        	       0x196119278 -[_NSTrackingAreaAKManager _invalidateCursorRectsForViewsWithNoTrackingAreas:] + 192
3   AppKit                        	       0x196119278 -[_NSTrackingAreaAKManager _invalidateCursorRectsForViewsWithNoTrackingAreas:] + 192
4   AppKit                        	       0x196119278 -[_NSTrackingAreaAKManager _invalidateCursorRectsForViewsWithNoTrackingAreas:] + 192
5   AppKit                        	       0x196119278 -[_NSTrackingAreaAKManager _invalidateCursorRectsForViewsWithNoTrackingAreas:] + 192
6   AppKit                        	       0x196119278 -[_NSTrackingAreaAKManager _invalidateCursorRectsForViewsWithNoTrackingAreas:] + 192
7   AppKit                        	       0x196119278 -[_NSTrackingAreaAKManager _invalidateCursorRectsForViewsWithNoTrackingAreas:] + 192
8   AppKit                        	       0x19611934c -[_NSTrackingAreaAKManager _invalidateCursorRectsForViewsWithNoTrackingAreas] + 100
9   AppKit                        	       0x1961193f8 -[_NSTrackingAreaAKManager invalidateCursorRectsForView:] + 56
10  AppKit                        	       0x195ba4a20 -[NSView removeFromSuperview] + 144
11  Sensors                       	       0x1049ff2fc 0x1049e4000 + 111356
12  Sensors                       	       0x1049e7790 0x1049e4000 + 14224
13  Sensors                       	       0x104a0d800 0x1049e4000 + 169984
14  Sensors                       	       0x104a0d854 0x1049e4000 + 170068
15  AppKit                        	       0x195d112f0 -[NSApplication(NSResponder) sendAction:to:from:] + 440
16  AppKit                        	       0x195d11108 -[NSControl sendAction:to:] + 72
17  AppKit                        	       0x195d0e324 NSControlTrackMouse + 1168
18  AppKit                        	       0x195d0d380 -[NSControl mouseDown:] + 852
19  AppKit                        	       0x195d0bce0 -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:] + 3420
20  AppKit                        	       0x195c958f0 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 1984
21  AppKit                        	       0x195c94f5c -[NSWindow(NSEventRouting) sendEvent:] + 284
22  AppKit                        	       0x195c94094 -[NSApplication(NSEvent) sendEvent:] + 1920
23  AppKit                        	       0x195ee6144 -[NSApplication _handleEvent:] + 60
24  AppKit                        	       0x195b5bc14 -[NSApplication run] + 500
25  Stats                         	       0x1049287d4 0x104924000 + 18388
26  dyld                          	       0x192507e50 start + 2544

Thread 1:
0   libsystem_pthread.dylib       	       0x19282ce18 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	       0x19282ce18 start_wqthread + 0

Thread 3:
[... 295 lines cut ...]
      },
      "deploymentId" : 240000040
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro18,1, BootROM 8419.41.10, proc 10:8:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Samsung
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Sep  3 2022 02:35:52 version 20.10.965.9.8.7.129 FWID 01-b0e84a9b
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: MagSafe Charging Case
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: Laptop, Apple Inc.
Thunderbolt Bus: Laptop, Apple Inc.
Thunderbolt Bus: Laptop, Apple Inc.
```

closed 2022-11-15 by exelban