exelban/stats #1173: Authentication needed for every fan RPM change

LostInCompilation, 2022-11-14:
**Describe the bug**
Since about 3 or 4 releases Stats asks me for a password every time I change the fan RPM in manual mode. Before I just needed to enter the password once when switching to manual fan mode and was able to change the RPM without re-authentication.

**Details:**
 - Device: MacBook Pro 2021 16" (M1 Pro)
 - macOS: macOs 13.0.1
 - Application version: 2.7.43 (Bug present since at least 2.7.41)

exelban, 2022-11-14:
Hi. Thanks for the feedback.
Please try to install 2.7.40 and check if it will ask for a password every time.

exelban, 2022-11-15:
Looks like it's similar to the #1166

Moulick, 2022-11-15:
Yep, this is exactly what I experienced.

exelban, 2022-11-15:
It's a very interesting bug. Because if I run the app via xcode it works normally. But after the build, I issue the same bug.

Moulick, 2022-11-15:
iStat menus is able to do control the fans just fine. The same version was able to control the fans in both Montrey and Ventura

exelban, 2022-11-15:
Yeap, I can confirm that it issues only new macOS 13.0.1

exelban, 2022-11-15:
Ok, I figured out why it does not work. The problem is the way how I ask for permission. I do that throw apple script (yeah, it's not the best way, but it works). And looks like this way stops working.
Looks like I need to implement an authorization service to make it work.

If it's important for you to control the fans you can run:
```
sudo /usr/sbin/chown root:admin /Applications/Stats.app/Contents/Resources/smc
sudo /bin/chmod 6555 /Applications/Stats.app/Contents/Resources/smc
```

After that Stats could control the fans.

PS: I'm not sure that it will be fixed this week, but I will try to do that.

Moulick, 2022-11-16:
yep, after the above it was able to control the speed but still it asked for the password 6 times

Moulick, 2022-11-20:
I just tried the 2.7.45 release. It asked for the password 6 times, asking to install helper, and still did not change the fan speed :)

exelban, 2022-11-20:
Yeap, notice that. I revoke the last update for now.

exelban, 2022-11-20:
Ok, I have updated the release. There was a problem with signing the helper app. Now it works fine. And fans could be controlled.

closed 2022-11-20 by exelban

Moulick, 2022-11-20:
Hmm, it works, but still asked for the password 4 times.

exelban, 2022-11-20:
Do you have 4 fans and enabled fans sync?

Moulick, 2022-11-20:
I have 2 fans with sync enabled

exelban, 2022-11-20:
Yeap, and I suppose you press the max speed button?

Moulick, 2022-11-20:
I pressed that snowflake-looking button on the extreme right. Dunno if that's full speed or 0 speed

exelban, 2022-11-20:
Yeap, it's a reason why you have been asked 4 times: 2 times for mode change, and after 2 times for speed change.

You could normally deny after the first time because the helper is already installed. I will fix this when will have time to rewrite these views. It's not critical and now for password Stats will ask only when first time manipulating the fans.